whatotter/pwnhyve #4: ADD U2F KEY HID support

dedsecproyt, 2023-03-11:
I need U2F a function (like in the frimware flipperzero) for logging into accounts and windows

nototter, 2023-03-14:
> I need U2F a function (like in the frimware flipperzero) for logging into accounts and windows

nototter, 2023-03-14:
didnt mean to close my bad

do you mean U2F key as in a yubikey? unless you want to replicate accounts n stuff, i dont see why this would be a good addition to the project - i can try to add it though, but until then i feel like making your own plugin to do what you want is a better idea

closed 2023-03-19 by nototter

dedsecproyt, 2023-03-24:
You can use [U2F-KEY HID Repo](https://github.com/flipperdevices/flipperzero-firmware/tree/dev/applications/main/u2f)  from FlipperZero code

GodlySchnoz, 2024-01-15:
only saw this now, might be worthwile to look at [This repo for U2F](https://github.com/Crystalix007/U2FDevice) or [this one for FIDO2](https://github.com/mphi-rc/pi-zero-security-key) and adapting them based on needs if i have time after the middle of february i will take a look and might take a shot at it, shouldn't be too difficult but need to make sure of compatibility with the zero 2 as that's what i have'
Honestly i am wondering on the usefulness in an attack device, cloning keys should be almost impossible and absolutely impractical, i know of a 2021 side channel attack so for sure won't evem bother to implement such scripts.
Ig using it as a U2F key is useful but personally i prefer dedicated ones like yubikey expecially since accessing to your own accounts on victim pc for example might be extremely stupid, only option left would be gaining access to victim accounts and ig that's useful but can be done with normal keys, this might be cheaper ig but the repos i linked should fare well enough also the installation imo should be optional, that said if i get to it i might write a good old script to install it.