whatotter/pwnhyve #6: Need help?

90N45-d3v, 2023-03-23:
I haven't tested your project yet, but you wrote that you need help with the front-end. I would love to help you...

Can you tell me what exactly you need help with? You said the design was really ugly, so you already got the backend working? If it's just the HTML/CSS design or something like that, I could build it without having to setup my Raspberry again.

nototter, 2023-03-24:
yes, i believe it's just the front end that needs work - back end still needs a little work but once thats done i will gladly take your help
do you have any way to directly contact you? telegram, discord, etc.
thank you so much btw

90N45-d3v, 2023-03-24:
Sure, you can hit me on Discord: 90N45#1983
I think with this, you can close the issue 👍

closed 2023-03-25 by nototter