whatotter/pwnhyve #11: Unable to locate package wiringpi

servebeer, 2023-08-21:
Initial research shows that this package has been deprecated...

whatotter, 2023-08-26:
oooh u are right, pwnhyve seems to work without it - try removing the package from the apt command, and see if it still works

servebeer, 2023-08-26:
I was able to locate the last revision of the package in an archive, so I just installed it and everything is working fine.

whatotter, 2023-08-27:
woohoo 🎉 until i push the next update, i can see if wiringpi isn't needed anymore and can maybe remove it from the installation page - thank you for letting me know though, much appreciated 👍

also, please let me know if you have any issues, since i cant diagnose all of them myself

servebeer, 2023-08-28:
Thanks for the update. The only issue I am having right now is with the directional controls on the display. When I press down/up, the selection moves 3 lines instead of just one. I am sure there is a setting for this, but I have not found it yet.

whatotter, 2023-08-29:
there isnt a setting for that, is it occasionally or all the time? it might be ur display, but i can definitely push a fix when i get back home if it isn't

servebeer, 2023-08-30:
It is happening all of the time. I have reseated the display hoping that would help, but unfortunately it does not.

whatotter, 2023-08-30:
do you have any other pihats? can you confirm your soldering? i can't seem to reproduce the issue
if you can, please take a video of it - it could be like 5 seconds long, i just want to see what your doing is correct

also, unrelated, but your my first ACTUAL issue - congrats 🎉

servebeer, 2023-08-30:
The board looks good (I took a pic of it). I also made a short video of the behavior.

[image]

https://github.com/whatotter/pwnhyve/assets/8514341/b26f3078-9f1a-43da-967a-13bc9ed1537e

servebeer, 2023-08-30:
The board looks good (I took a pic of it). I also made a short video of the behavior.

[image]

https://github.com/whatotter/pwnhyve/assets/8514341/b26f3078-9f1a-43da-967a-13bc9ed1537e

Also, I just noticed that this behavior only occurs on the main menu. It scrolls one line at a time on the sub menus that I have tried.

whatotter, 2023-08-30:
ohhh, so it seems like an issue with key functions - i should be able attempt to do a fix in about 3ish hours

thank you very much 🙏

whatotter, 2023-09-01:
procrastination has unfortunately took me under and so did school work
i have the fix (i think) but i got bored and started messing with the wifi plugin and might've messed it up

oops

servebeer, 2023-09-01:
lol

No problem... this is a hobby for me. I really appreciate your time and effort!

whatotter, 2023-09-01:
okay, i pushed only the changes for main.py, which should fix the issue - please update me if it works after, when you can of course

servebeer, 2023-09-02:
Bad news... now the directional controller doesn't work at all :-(

whatotter, 2023-09-03:
uh oh, do other buttons work? like the ones on the right? this might be an issue with the joystick, which might mean you have to resolder a new one, or something like that

i really hope not, but it works perfect on my end

servebeer, 2023-09-05:
So I decided to start all the way over from scratch, but with the same results. My only guess is that I am using a different LCD panel than you? I am using a Waveshare 1.3inch OLED Hat.

[image]

whatotter, 2023-09-06:
seems to be the same oled hat as me, ina bit i can make a test script that checks if it's your display

servebeer, 2023-09-06:
I appreciate all the time that u r putting into this! There is no hurry.

whatotter, 2023-09-07:
okay, since schoolwork has grabbed me by the balls and i am also now unfortunately sick, i can't make a script

[if you go to waveshare's site, there is an example script that checks if the buttons work](https://www.waveshare.com/wiki/1.3inch_OLED_HAT#Download_Example), and you can check your terminal if the buttons come up 3 times when you press the directional buttons

to minimize the variability, use the python3 version of the script just in case the C version some how fixes your issue

whatotter, 2023-09-09:
i was unfortunately diagnosed with covid but that means i have more time to work on it and still have time to rest, as i now have an entire week to myself

i believe i have the fix in my hands (hopefully)

servebeer, 2023-09-09:
Man, I hope that it is a mild case. Everyone I know who has gotten it this year says it is like having a cold.

whatotter, 2023-09-09:
seems like it is a mild case, i feel the same as i always have
anyways, i pushed a possible fix - when you can, please try it

servebeer, 2023-09-09:
Glad to hear that it isnt bad. I will test the update tomorrow and let you know.

servebeer, 2023-09-09:
My AT&T has Internet has goon out and they wont be here to look at it until Monday afternoon 😞

whatotter, 2023-09-20:
if your internet is back up, here's a quick reminder

servebeer, 2023-09-20:
Sorry about that... yea AT&T found the fiber break and got it fixed. Unfortunately I had to go back into the office and I have been swamped since. I promise I will take a look at this tonight and post something.

servebeer, 2023-09-20:
Good deal, The menu is now moving only 1 line at a time!

whatotter, 2023-09-21:
> Sorry about that... yea AT&T found the fiber break and got it fixed. Unfortunately I had to go back into the office and I have been swamped since. I promise I will take a look at this tonight and post something.

definitely fine, i have the same with school

> Good deal, The menu is now moving only 1 line at a time!
woohoo 🎉🎉🎉 so that fixes the issue, which means i can close this issue now - thank you so much for your help! if you have any other issues, please let me know, as ill be more than happy to fix them

closed 2023-09-21 by whatotter