whatotter/pwnhyve #21: 1.44inch lcd screen issue

sir-francisdrake, 2023-12-30:
Really trying hard to get this to work. look like everything is good except when the script is running I cannot get it to display on my screen when starting up. (screen is just white). Here is the screen I am trying to use: https://www.waveshare.com/wiki/1.44inch_LCD_HAT#Demo. Sizing is a bit off but it appears everything else should be the same. Any help or advice would be greatly appreciated.

whatotter, 2023-12-30:
the LCD hat is different from the OLED hat, the python script only support SH1106 OLED drivers, not LCD drivers
i want to add support to it but it's going to be further down the road, so as of right now your only options are:

- add support for it yourself (i don't recommend)
- use the OLED hat
- wait for me to add support for it (will take a while!)

sir-francisdrake, 2023-12-30:
Thank you for a quick response! After some fiddling I may just grab the other screen. Seems like the better  option for the time I have to spend on this. Keep up the good work!

closed 2023-12-30 by sir-francisdrake