whatotter/pwnhyve #23: Problem with the wiki

GodlySchnoz, 2024-01-15:
The wiki references pwnhyvessh that was removed in commit 1be919f4667a1aca60caf96e9b7f00d645b1853f

GodlySchnoz, 2024-01-15:
And also referencing #11 wiringpi is deprecated so should either include the legacy library, remove it completely or substitute it
[this is an unofficial mirror with some added hardware support with various wrappers one of which is in python](https://github.com/WiringPi/WiringPi) also since the project is in python [something like gpiozero might be worthwile to look at](https://gpiozero.readthedocs.io/en/stable/)

whatotter, 2024-01-16:
thank you for telling me this, i will edit it asap

will not last long though, as pwnhyve is undergoing a full rewrite with cython/go/c++ because of how slow it is with basic python, especially basic SPI writing

plus, pwnhyve doesn't need wiringpi afaik, but i will work on that

whatotter, 2024-01-17:
until i push the new rewrite, i advise not working on any plugins since it will break support for them - i should finish in about a week or so :)
plus, i might just work on it myself and you can contribute to it

GodlySchnoz, 2024-01-17:
yea that's not a problem at all, it's by no means something hard to do, the exploit itself is quite basic as it just needs to randomize the last 3 sets of bytes to randomize the device identification and the other ones are kind of set in stone as they describe averything else also it the rewrite is done in less than 3 weeks for sure i won't be able to work on it comes out as i have my uni exams.

whatotter, 2024-02-25:
closed due to lack of activity

closed 2024-02-25 by whatotter