whatotter/pwnhyve #28: Pwngotchi Module gets stuck on "missed": "(☼/\\☼)",

Laughable33, 2024-02-20:
As the title says, when I load up the pwngotchi module, it boots up and gets stuck on the "missed": "(☼/\\☼)", face. If you could direct me to where the logs are stored I would send those also.

whatotter, 2024-02-20:
the logs aren't saved to a place as of yet to prevent storage from getting bloated - you'd need to ssh into the pi, kill the currently running pwnhyve instance, and then go to the folder where you have pwnhyve and do `sudo python3 main.py`, then it'll work as normal and you'll have logs

so:
1. kill the currently running pwnhyve instance (`sudo pkill -f main.py` might work but i don't recommend it)
2. go to your folder where you have pwnhyve saved
3. `sudo python3 main.py`

Laughable33, 2024-02-24:
Here are the logs 

['beaconSpam', 'rssiReader', 'setInterfaces', 'pwnagotchi', 'evilPortal']
13
------------------
pwnagotchi
['wifi.py', 'evilportal.py', '__pycache__']
whitelist: 

After that it just shows the face as described before.

whatotter, 2024-02-24:
okay thats my fault, i never really did logging on the pwnagotchi module - if you could please update pwnhyve with the most recent commit and try it again that'd be sick, should be more verbose

once you do that and if it doesn't fix, show the last 50ish lines of your `sudo dmesg` alongside the error it gives

Laughable33, 2024-02-24:
I love the new UI! 

Here is the logs from running the python 

[-256]
[PWNAGOTCHI] whitelist: 
[WIFI] wlan0's mode: Managed
[WIFI] attempting to set wlan0 in monitor mode

Here is the logs from sudo dmesg

[    8.426383] bcm2835-isp bcm2835-isp: Register output node 0 with media controller
[    8.430712] bcm2835-isp bcm2835-isp: Register capture node 1 with media controller
[    8.471767] bcm2835-isp bcm2835-isp: Register capture node 2 with media controller
[    8.476066] bcm2835-isp bcm2835-isp: Register capture node 3 with media controller
[    8.501833] bcm2835-isp bcm2835-isp: Loaded V4L2 bcm2835-isp
[    8.556132] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    8.643580] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    8.722139] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[    8.736504] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436s-sdio for chip BCM43430/1
[    8.742373] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43436s-sdio.raspberrypi,model-zero-2-w.bin failed with error -2
[    8.748430] usbcore: registered new interface driver brcmfmac
[    8.924862] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436s-sdio for chip BCM43430/1
[    8.929097] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436s-sdio for chip BCM43430/1
[    8.933343] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[    8.941819] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Feb  5 2021 12:49:07 version 7.45.96 (r745790) FWID 01-71817851 es7
[   13.326812] uart-pl011 3f201000.serial: no DMA platform data
[   15.036007] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   15.586646] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   16.507664] Bluetooth: Core ver 2.22
[   16.510027] NET: Registered PF_BLUETOOTH protocol family
[   16.512312] Bluetooth: HCI device and connection manager initialized
[   16.514629] Bluetooth: HCI socket layer initialized
[   16.516901] Bluetooth: L2CAP socket layer initialized
[   16.519150] Bluetooth: SCO socket layer initialized
[   16.537172] Bluetooth: HCI UART driver ver 2.3
[   16.539378] Bluetooth: HCI UART protocol H4 registered
[   16.541595] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   16.543977] Bluetooth: HCI UART protocol Broadcom registered
[   17.127278] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   17.129264] Bluetooth: BNEP filters: protocol multicast
[   17.131152] Bluetooth: BNEP socket layer initialized
[   17.161267] NET: Registered PF_ALG protocol family
[   17.198018] cryptd: max_cpu_qlen set to 1000
[   17.360228] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   18.124822] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   26.403669] systemd-journald[174]: File /var/log/journal/d578738f51fe469ba11df9163e7dfcd4/user-1000.journal corrupted or uncleanly shut down, renaming and replacing.
[   27.753221] Bluetooth: RFCOMM TTY layer initialized
[   27.753374] Bluetooth: RFCOMM socket layer initialized
[   27.753509] Bluetooth: RFCOMM ver 1.11
[   32.471772] cam-dummy-reg: disabling

whatotter, 2024-02-24:
> I love the new UI!

thank you, it was based off of the flipper zero's UI :)

does the pwnagotchi module cause the pi to crash? or is it just stuck? dmesg doesn't show that there was an attempt to put the wifi interface into monitor mode either, so it might be stuck trying to put it into monitor mode

1. are you running pwnhyve with sudo rights?
2. does wlan0 exist and supports being put into monitor mode?
3. is wlan0 already in monitor mode?

if you also can, try putting wlan0 in monitor mode before running pwnhyve to see if it just gets stuck there (`sudo airmon-ng start wlan0`)

Laughable33, 2024-02-25:
1. Yes it is being run with sudo rights

Running sudo airmon-ng start wlan0 crashes the pi

┌──(root㉿kali-raspberry-pi-zero-2-w)-[~/pwnhyve]
└─# sudo airmon-ng start wlan0

Found 2 processes that could cause trouble.
Kill them using 'airmon-ng check kill' before putting
the card in monitor mode, they will interfere by changing channels
and sometimes putting the interface back in managed mode

  PID Name
  553 NetworkManager
  583 wpa_supplicant

PHY     Interface       Driver          Chipset

phy0    wlan0           brcmfmac        Broadcom 43430

whatotter, 2024-02-25:
are you sure it crashes your pi and doesn't just kill your ssh terminal?

if it does actually crash your pi, something is wrong in the kernel and i can't help much with that as I've never had that issue

Laughable33, 2024-02-25:
It kills ssh but also freezes the pwnhyve screen.

whatotter, 2024-02-25:
if you can, can you connect the pi to a monitor and keyboard and run pwnhyve through there? it'll show if it's just the wlan card is dying or if it's the actual pi, plus it should do logs if it kernel panics so you can fix that

whatotter, 2024-08-06:
closing due to inactivity (6 months!)

closed 2024-08-06 by whatotter