whatotter/pwnhyve #29: Flashable (Updatable) .bin files in the future?

TeqTinkerer, 2024-02-21:
Hi! Big love for this project! I was wondering if for making things easier for installation and updates, if it's possible in the future to see .bin files to flash on sd card and install? :)

whatotter, 2024-02-21:
it's definitely in the roadmap, and i thought of it a loong time ago, but i thought not much people would really care if they install it or if it's a bin file, and to put less work on myself by doing extra stuff and setting more time to work more on the project, i just set it aside.

i feel like at where we are right now in the project, i don't think it'd be much useful considering you'd still need to setup wifi for it to auto update, so you'd still be stuck at the terminal for a bit

still if you'd like to work on it, you definitely could and i'll definitely merge it to the project (assuming it does work), otherwise it'll take a while for it to arrive to the project :)

whatotter, 2025-07-01:
this is a thing now, so, closing

closed 2025-07-01 by whatotter