whatotter/pwnhyve #30: Run from a headless pi

igeometrix, 2024-02-24:
I don't see in the documentation how exactly I can run this from a headless pi. How do configure it and then how do I actually get to control or select any of the modules. I basically have it setup on my pi to auto run like the documentation shows but can I say, ssh into the pi and get to a menu of sorts or am I going about it the wrong way?

whatotter, 2024-02-24:
running headless is possible, but in the `config.toml` [VNC must be enabled](https://github.com/whatotter/pwnhyve/blob/main/config.toml#L12-L14). once that's done, you need to clone the pwnhyve repo to your PC, and then go to the vnc folder, then do `python3 stream.py (ip address of your pwnhyve pi)`.

if you want to do wifi hacking while using vnc you need a second wifi adapter, because when pwnhyve puts wlan0 into monitor mode it'll kill your session and you won't be able to get it back

VNC is also still experimental, report any issues you have on it :)

igeometrix, 2024-02-25:
OK thank you, I got it going but sometimes the interface has info available and sometimes its just blank after I connect the IP. Maybe I should reinstall on my pi since I see you've been updating quite a bit lately. Is there a good way to just update a current install on my pi?

igeometrix, 2024-02-25:
Main issue is that even when it connects and I have some options on the interface.. I only see a few and say I go into /Hotplug.. it then fails on the raspi side and disconnects from the stream.py. Do you have any input on this or do you think the update as I said above could fix all this? Also curious where the hoaxshell rev shell setup is at within the menus?

whatotter, 2024-02-25:
> OK thank you, I got it going but sometimes the interface has info available and sometimes its just blank after I connect the IP. Maybe I should reinstall on my pi since I see you've been updating quite a bit lately. Is there a good way to just update a current install on my pi? 

as for right now, the systeminfo util is kinda broken and other plugins are being prioritised

and as for updating, you'd have to ssh into the pi and git clone the new repository - an auto update is in the works but redundancy is something highly prioritised too, so even if the update fails for some reason it should still work

whatotter, 2024-02-25:
> Main issue is that even when it connects and I have some options on the interface.. I only see a few and say I go into /Hotplug.. it then fails on the raspi side and disconnects from the stream.py. [...]

as for the raspi issue, I've never seen that happen and i use the vnc plugin almost daily, try git cloning the new repository on your pi and then show the logs here 🙂

the hoax shell plugin is being reworked at the moment but it should be under /Hotplug afaik, if not then it should be implemented next update (if I'm not lazy)

igeometrix, 2024-02-25:
Here's some screens with the errors im getting.. some from my machine and some from the pi (ssh)
[image]
[image]
[image]
[image]

whatotter, 2024-02-25:
try updating the pi and then making sure you went [successfully through the badusb setup](https://github.com/whatotter/pwnhyve/wiki/installing#usb-module-run-on-boot), it might be because /dev/hidg0 doesn't exist, weird

whatotter, 2024-02-25:
is the cable your using data? did you reboot the pi after installing? if you can, try a different port too

another thing you can try is doing `sudo /bin/pwnhyveusb` (or wherever your usb init module is) while the pi is connected

wvpul, 2024-06-22:
Hi, i have been getting an error everytime i try to do "python stream.py my pi ip" 
[image]

whatotter, 2024-07-08:
closed due to inactivity, was fixed in the discord server

closed 2024-07-08 by whatotter