whatotter/pwnhyve #35: run without oled?

someguy1412, 2024-04-07:
if its possible then how?

someguy1412, 2024-04-07:
it runs fine now but i see zero modules
[image]

chufo4, 2024-04-07:
this is my alt and it has modules but they dont work

whatotter, 2024-04-08:
sorry, the vnc module is still in heavy development atm so it might be very buggy

the controls are (iirc) the arrow keys as the joystick, 1,2,3 as the right side keys, and the forward slash key as joystick down

you could also click on the buttons with your mouse on the ui as if your using the real display

I'll switch the module to websockets eventually, so there won't be anymore socket issues

someguy1412, 2024-04-08:
oh thanks, yeah i cant buy any oleds because none are in my region, so i said maybe i could use python for the vnc and get the display but yeah its heavily buggy sadly

closed 2024-04-08 by someguy1412