whatotter/pwnhyve #36: NOT Issue, it's a question about HID

quesodipesto, 2024-04-10:
Hi mate, first of all, thank for develop this fuc**** amazing hacking tool.
I have one question. As i think that you know, in P4wnp1 A.L.O.A, we are able to use the HID with javascript code.

Question 1: Here que can use 1:1 the P4wnp1 HID scripts? we have to modify firstly the HID script (maybe transform in to RubberDucky lenguage)? 

Question 2: We can move the mouse with HID commands?

Thanks a lot!!! 

PD: This project have Discord channel?

whatotter, 2024-04-11:
> Hi mate, first of all, thank for develop this fuc**** amazing hacking tool.  

thank you very very much, makes me happy to see people enjoying my dumb side projects :)

> Here que can use 1:1 the P4wnp1 HID scripts? we have to modify firstly the HID script (maybe transform in to RubberDucky lenguage)?  

unfortunately no, the HID scripts for the p4wnp1 are ***not*** cross compatible. i had the choice whether to pick rubber ducky vs p4wnp1's HID scripts, and since rubber ducky scripts are easier to write, have tons and tons of possibilities, and a repo is already made for them, i picked those instead :)

> Question 2: We can move the mouse with HID commands?

yes, you can use `MOVE x y` to move the mouse in absolute coordinates, e.g. `MOVE 10 10` would move the mouse 10 pixels to the right and 10 pixels up, and `MOVE -10 -10` would move the mouse 10 pixels down, and 10 pixels to the left

> PD: This project have Discord channel?
yes, https://discord.gg/HdPKgHQj

quesodipesto, 2024-04-13:
> thank you very very much, makes me happy to see people enjoying my dumb side projects :)
>
>
> unfortunately no, the HID scripts for the p4wnp1 are _**not**_ cross compatible. i had the choice whether [...]

Ok :D thanks again for your work! 

I have problems with your discord link, the link does not seem to work.

On the other hand, is the tool able to simulate an HDD + HID? I want to be able to launch a code, which allows me to extract a file from the raspberry and copy it directly to the local directory of the target and then execute it.

Thanks mate!

whatotter, 2024-04-13:
yes, it is able to emulate a usb mass storage device - https://github.com/whatotter/pwnhyve/wiki/installing#installing  
should show up on plug in too

and sorry, the discord link does seem to be expired, https://discord.gg/azZznGrWDk should work

closed 2024-05-17 by whatotter