whatotter/pwnhyve #37: dogwater suggestion

chufo4, 2024-04-12:
as the pwnagotchi and p4wnp1 aloa already use web uis could you implement them instead of vnc? ofcourse this is significantly harder tho. 

title says alot about this

chufo4, 2024-04-12:
ps: if you release these:
with the shim (WIP, not released to public yet)

RF hacking (rolljam, sniffing, jamming, replay)
IR hacking (replay, jamming)
wireless charging
pin fuzzing (SPI, IIC/I2C, UART)
nfc space constrained and also very hard to make

do we also need some extra stuff since im not sure if  a normal pi zero w/2 w could do all that standalone

whatotter, 2024-04-13:
> as the pwnagotchi and p4wnp1 aloa already use web uis could you implement them instead of vnc?

i do plan to, but since we can't use wifi (because of deauthing), we'd have to use bluetooth PAN, which would then mean we can't use sourapple attacks (which is being worked on), so its one or the other

> do we also need some extra stuff since im not sure if a normal pi zero w/2 w could do all that standalone

correct, it'll be as a pcb shim + parts you'd have to solder on, you can do pin fuzzing without all that though

chufo4, 2024-04-13:
alright for now I will stick to my pi zero 2w with p4wnp1 aloa hmm I'm not familiar with those other parts too well tho. so is there any way to use this without display? maybe I'm too dumb for this shit?  (vnc excluded unless I fix my errors with no modules showing up)

btw the p4wnp1 aloa already has kali could I just build it on top of that? I'm tired of resetting my SD card every time
and I'm pretty sure berryboot doesn't work

(bro can't find another SD card in stock)

chufo4, 2024-04-13:
forgot to mention, any other displays I can't find the "waveshare 1.3inch OLED  HAT" or similar anywhere I live in austria so any help would be very helpful

whatotter, 2024-04-14:
> could I just build this on top of that?

unfortunately, no, p4wnp1 would overlap with pwnhyve and bug out a LOT, you'd have to install a clean version of kali linux on the pi, making you have to reset the sd card

>  (vnc excluded unless I fix my errors with no modules showing up)

not your fault, i'll work on that :) i'll focus on that mainly

> I'm not familiar with those other parts too well tho

dont worry about that either, the only one(s) you really need is the pi and the display

> "waveshare 1.3inch OLED HAT" or similar anywhere I live in austria so any help would be very helpful
what do you mean? like you can't find an austrian seller?

chufo4, 2024-04-14:
for the last one I found a 1.44 inc h waveshare one but none 1.3inch will it work?

someguy1412, 2024-04-14:
ps: abiut the ir hacking/blasting thing what parts does it need

chufo4, 2024-04-14:
nvm

closed 2024-04-14 by chufo4