whatotter/pwnhyve #38: E: Package 'python3-smbus' has no installation candidate E: Unable to locate package bettercap

bladedevoff, 2024-06-26:
Hi, guys! Can you help me with this problem? Raspberry Pi Zero 2W.

Package python3-smbus is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'python3-smbus' has no installation candidate
E: Unable to locate package bettercap

whatotter, 2024-06-29:
are you on raspbian? or on kali?
python3-smbus isn't needed unless your using the display in I2C, which isn't really supported (yet)  

the core issue seems to be bettercap which isn't needed if you don't use the wifi module, which is broken on the raspi2w (re4son kernel is broken or something) so if you can't install it you should be fine

try running an `apt update` and trying again

bladedevoff, 2024-06-29:
I'm using Kali, can i buy external usb adapter for bettercap? Or install fixed kernel maybe?

whatotter, 2024-07-01:
yes you can use an external wifi adapter and that'd work fine, then you wouldn't need a new kernel

whatotter, 2024-08-06:
closing due to inactivity

closed 2024-08-06 by whatotter