whatotter/pwnhyve #43: Service will not run

servebeer, 2024-10-04:
I just completed an install (with no errors), I the rebooted thePI (still no errors), but when I do a "sudo systemctrl status pwnhyve.service" the service is not running:

┌──(kali㉿kali-raspberry-pi-zero-2-w)-[~]
└─$ sudo systemctl status pwnhyve.service
× pwnhyve.service - pwnhyve startup
     Loaded: loaded (/etc/systemd/system/pwnhyve.service; enabled; preset: disabled)
     Active: failed (Result: exit-code) since Fri 2024-10-04 13:32:56 UTC; 7s ago
   Duration: 3.546s
 Invocation: e8bbc5c170914164ae2775facb4c6853
    Process: 954 ExecStart=/bin/pwnhyveStart (code=exited, status=139)
   Main PID: 954 (code=exited, status=139)
        CPU: 1.752s

Oct 04 13:32:56 kali-raspberry-pi-zero-2-w pwnhyveStart[957]:           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Oct 04 13:32:56 kali-raspberry-pi-zero-2-w pwnhyveStart[957]:   File "/home/kali/pwnhyve/core/screens/sh1106.py", line 114, i>
Oct 04 13:32:56 kali-raspberry-pi-zero-2-w pwnhyveStart[957]:     c = self.checkIfKey(debounce=False, key=x)
Oct 04 13:32:56 kali-raspberry-pi-zero-2-w pwnhyveStart[957]:         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Oct 04 13:32:56 kali-raspberry-pi-zero-2-w pwnhyveStart[957]: TypeError: DisplayDriver.checkIfKey() got an unexpected keyword>
Oct 04 13:32:56 kali-raspberry-pi-zero-2-w sudo[956]: pam_unix(sudo:session): session closed for user root
Oct 04 13:32:56 kali-raspberry-pi-zero-2-w pwnhyveStart[954]: Segmentation fault
Oct 04 13:32:56 kali-raspberry-pi-zero-2-w systemd[1]: pwnhyve.service: Main process exited, code=exited, status=139/n/a
Oct 04 13:32:56 kali-raspberry-pi-zero-2-w systemd[1]: pwnhyve.service: Failed with result 'exit-code'.
Oct 04 13:32:56 kali-raspberry-pi-zero-2-w systemd[1]: pwnhyve.service: Consumed 1.752s CPU time.
lines 1-19/19 (END)

whatotter, 2024-10-04:
can you try to run it manually through ssh? go to the pwnhyve folder and do `sudo python3 main.py`, and add the output here

servebeer, 2024-10-04:
┌──(kali㉿kali-raspberry-pi-zero-2-w)-[~/pwnhyve]
└─$ sudo python3 main.py   
sh1106.py
[SL] attempting to load display driver "sh1106.py"
core.screens.sh1106
[VNC] READY FOR CONNECTIONS.
[SL] loaded
[DISPLAY] initalizing display driver...[DISPLAY] initalizing display driver.../ [GPIO] initalizing GPIO pins...
| [GPIO] initializing GPIO pin 6...initalized
| [GPIO] initializing GPIO pin 19...initalized
| [GPIO] initializing GPIO pin 5...initalized
| [GPIO] initializing GPIO pin 26...initalized
| [GPIO] initializing GPIO pin 13...initalized
| [GPIO] initializing GPIO pin 21...initalized
| [GPIO] initializing GPIO pin 20...initalized
| [GPIO] initializing GPIO pin 16...initalized
\ [GPIO] done initalizing GPIO pins.

initalized.

[PLUGINS] initializing plugins and menus...[PL] loading plugin "plugins.onIdle"
{}
[PL] loading class "onIdle::Plugin"
initalized.
[MENU] READY. HACK THE PLANET!
Traceback (most recent call last):
  File "/home/kali/pwnhyve/main.py", line 129, in <module>
    key = disp.gui.menu(pnd, disableBack=currentDirectory=="", icons=plugins.icons)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/pwnhyve/menus/flipper.py", line 68, in menu
    key = self.disp.waitForKey(debounce=True)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/pwnhyve/core/screens/sh1106.py", line 114, in waitForKey
    c = self.checkIfKey(debounce=False, key=x)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: DisplayDriver.checkIfKey() got an unexpected keyword argument 'debounce'

whatotter, 2024-10-04:
weird, i don't get that issue - what version of python?

servebeer, 2024-10-07:
So, it helps to be smarter than the equipment that you are trying to use. I spent all weekend trying to figure this out. This morning, I happened to glance at the board and I saw some GPIO pins that I should not be seeing. Sure enough, I had not seated the display properly. I turned everything off, re-seated everything "properly" and turned it back on. Guess what, everything is working fine now. DUH

closed 2024-10-07 by servebeer