whatotter/pwnhyve #46: Selecting Menus does nothing

servebeer, 2024-10-10:
So, from a base install, I see all of the sub menus: (/Analyzer, /Debug, etc) if I try to select a sub menu (arrow down to highlight it, and then press right on the directional stick, nothing happens. I have shelled into the Pi and verified that the sub menu directories have the corresponding .py files. Am I missing something here?

v/r
Jon

servebeer, 2024-10-10:
@whatotter I am using a Wavwshare 1.3" OLED hat. The joystick on it only offers: up, down, left, and right. There are 3 keys. Key 1 doesn't appear to do anything, Key 2 flips the display 180 degrees, and Key 3 doesn't appear to do anything. I have tried pushing straight down on the joystick, but that doesn't do anything either.

whatotter, 2024-10-10:
whaat, is it an old version of the oled? my oled has left, right, down, up, and middle press for the joystick, alongside the 3 keys on the right

maybe debris?

whatotter, 2024-10-10:
fixed anyways in the newest commit

servebeer, 2024-10-10:
It is a REALLY old version of the board. I don't recall center-press ever working. I will go ahead shut it down and give it a good cleaning just in case.

whatotter, 2024-10-10:
if by chance it doesn't work, just git clone the newest version - this one has right-select support

servebeer, 2024-10-10:
> fixed anyways in the newest commit

Thanks @whatotter , the right-press on the joystick is now selecting the option.