whatotter/pwnhyve #49: Combine with @NightRang3r P4wn P1 for better version

sivabala21, 2024-11-07:
[NightRang3r](https://github.com/NightRang3r/P4wnP1-A.L.O.A.-Payloads)
NightRang3r has released a pretty good pi current image. Pwnhyve can be combined with NightRang3r, and this takes P4wn P1 aloa to the next level. I think this combination will be a good one.

shellnowasklater, 2025-03-08:
Is it hard to do this? I have tried my own shittygotchi code with p4p1 aloa before and it was quite hard to not make them shit the bed on each other.

whatotter, 2025-03-08:
not sure why issue was closed..

> Is it hard to do this? I have tried my own shittygotchi code with p4p1 aloa before and it was quite hard to not make them shit the bed on each other.

LOL it shouldn't be too hard at all, in the case of the payloads OP posted, all you'd need is to make a plugin with a parser

there's good docs (biased because i made them) [here](https://docs.pwnhyve.xyz/plugin-documentation/creating-your-own-plugin) to make a plugin for pwnhyve

whatotter, 2025-04-21:
i still have not found out what op wants me to do but i assume it's make more payloads, which is not a me issue

closed 2025-04-21 by whatotter