whatotter/pwnhyve #50: New Feature Idea : Fm Hijack

sivabala21, 2024-11-07:
I'd like to add a feature to this tool that would allow it to hijack FM stations. The Pi zero w can produce frequency modulation through its GPIO pin, so it should be able to accomplish this. The PiFm library could be used to implement this feature, but it uses the .wav format for broadcast. To use other formats such as mp3, sox, ffmpeg, or pygame could be used to transmit the file. The feature would allow the user to specify the format of the audio file to be broadcast. This would be a useful feature .

whatotter, 2024-11-07:
that's smart and might be possible with rpitx, but the cc1101 module that will implemented might interfere with it

sivabala21, 2024-11-08:
Theoretically, the Raspberry Pi Zero’s GPIO can generate frequencies within the 88.0 to 108.0 MHz range, so it should not interfere with sub-GHz frequencies, which operate around 433 or 315 MHz. However, to confirm, we need to test this practically. Even so, to prevent any potential interference, we should ensure that when sub-GHz emulation is active, the FM broadcast should be disabled.

whatotter, 2025-01-16:
working on this soon, should be implemented by next week hopefully

whatotter, 2025-04-21:
it's been 3 months but it's finally implemented !

<sub>yikes</sub>

closed 2025-04-21 by whatotter