whatotter/pwnhyve #52: OLED hat uses I2C or 4-SPI

TriDEntApollO, 2024-12-24:
So I am actually building my own 1.3inch OLED hat from scratch since its very expensive here, where I live so i was asking are u using the waveshare OLED hat with the 4-SPI protocol it comes preconfigured with or 3-SPI or I2C?? This is because i will have to build my own hat accordingly then. 

Thank you for your efforts and response in advance ;)

TriDEntApollO, 2024-12-24:
Also if u have used the 4-SPI interface then can u help me on how can i use I2C instead

whatotter, 2024-12-24:
at the moment (as of the most recent commit), the oled display uses 4spi - clock, miso and mosi, and chip select

which display are you trying to use for this? ssd1306 based, sh1106 based, etc.

TriDEntApollO, 2024-12-25:
I am using a SH1106 based I2C display

TriDEntApollO, 2024-12-25:
which files would i have to modify if i wanna use the I2C protocol instead?

whatotter, 2024-12-25:
> I am using a SH1106 based I2C display

SH1106 has support for 4spi, it's the display natively supported by this project in 128x64

whatotter, 2024-12-25:
> which files would i have to modify if i wanna use the I2C protocol instead?

https://github.com/whatotter/pwnhyve/tree/main/core/disp_drivers/SH1106

closed 2025-01-16 by whatotter