whatotter/pwnhyve #58: More detail on the Pcb/Shim

Tomomoto10, 2025-02-28:
Real interested in this project, have been looking at others but your featureset and usability look nice. Could you give any more info on the pcb shim? Pics, parts assembly?

whatotter, 2025-02-28:
PCB shim is in early beta stage, FYI you'd be the first one other than me to try it :)

it's a mish-mash of different PCB modules:
- [charge discharge module](https://www.amazon.com/MakerFocus-Discharge-Integrated-Charging-Protection/dp/B07PZT3ZW2)
- [CC1101 module](https://www.amazon.com/SHUATI-Frequency-Transceiver-Integrated-Industrial/dp/B0CQVMN5LM)
- some pin headers

[image]
(UV resin on the thin part of the screen because i'm a little heavy handed and clumsy, common shatter point (from experience))

[image]

here's the PCB shim with only the CC1101 and one side of the pin headers on, currently experimenting with the charge-discharge modules so that's not on, and a diode goes on D1 to prevent the pi from backfeeding the charge-discharge circuit, if you like to live risky you can just bridge it with solder or a wire

Tomomoto10, 2025-02-28:
Ah yes i see how it all goes now! Thanks for getting back so quick with good pics! 
Im really keen just waiting to get paid so i can order the bits.

Tomomoto10, 2025-02-28:
If you document all this well and it all works properly i recon you could take over pwnagotchi with this, That things is trivial compared.

whatotter, 2025-02-28:
> If you document all this well and it all works properly i recon you could take over pwnagotchi with this, That things is trivial compared.

i agree, but there is a lot of optimizations and things to fix before i can even think of that 😆 need to figure out cases aswell, and support for other 128x64 displays

will be waiting for your response on the shim, i do need testers for it

Tomomoto10, 2025-03-02:
This is my new autistic thought loop so ive pretty much only thought about this the last few days/nights. Have many thoughts and many questions 😂 will leave many for once ive got it all and set up. 
Can you overall explain shim and its pins, i get the c1101 is pinned to the appropriate pins on the pi.
But what are the 1-18 pins and those outward header pins? 
and whats the charge/discharge module for, how does this fit with pisugar?

whatotter, 2025-03-02:
haha no worries 😸 

the 1-18 pins are for external stuff, like how the flipper zero has add on boards, same thing here - or to connect to a breadboard, probes, etc.

the charge/discharge module was meant to be a BMS so you didn't have to have a bulky pisugar in the back of the pi - it'd use the space between the screen and the RPi for the battery management system, and you'd put the battery behind the RPi, making it way slimmer (you could still use a pisugar though)

I advise you **not** to order the shim PCB yet, i have some revisions to make that i just found out today, will edit asap and keep you updated

Tomomoto10, 2025-03-02:
Yeah i get it now, i did figure that wad what was happening looking at the traces but wasnt sure if you had some wizardry planned.. although i do see the IR 🫣

Tomomoto10, 2025-03-04:
You got a discord? Would be great to pick your brain about it all and share some ideas

whatotter, 2025-03-04:
yep there is a discord server, the invite link is at https://docs.pwnhyve.xyz

whatotter, 2025-04-21:
closing now that the pcb shim is out of alpha stage and into beta

closed 2025-04-21 by whatotter