whatotter/pwnhyve #62: Nothing shows on the screen

Deathlord8027, 2025-04-28:
Everything in the script seems to be working fine, except for the fact that nothing was showing on the screen. I checked the pins for damage and they seemed fine, just a little bent. I fixed that by using tweezers. I made sure that the screen was pushed all the way in, but still, nothing was posting. When I reran the code for the special SPI pins,  I noticed it said: "FATAL ERROR: Couldn't open "./core/install/spi-cs-extend.dts": No such file or directory
cp: cannot stat '3spi.dtbo': No such file or directory" Also, when I ran the requirements command, it said that it was an external enviorment, adding on that I had to create a virtual enviorment using python 3.

whatotter, 2025-04-28:
can you attach your `pwnhyve.log` here? it should be in the root directory of pwnhyve's folder

Deathlord8027, 2025-04-28:
Absoulute noob here, but it says '
pwnhyve.log': no such file or directory

Deathlord8027, 2025-04-28:
Also something else I noticed, when using the script for an externally managed eviorment(venv) it stated: error, not enough space. Is it possible to allocate more space for this?

whatotter, 2025-04-29:
> not enough space. Is it possible to allocate more space for this?

what size sd card are you using? minimum is 32gb

> Absoulute noob here, but it says '
pwnhyve.log': no such file or directory
when I run find

try `cat ~/pwnhyve/pwnhyve.log`  
if that doesn't work, try running the script manually: `sudo python3 main.py` and copy and paste the output here

> Also, when I ran the requirements command, it said that it was an external enviorment, adding on that I had to create a virtual enviorment using python 3.

this is a pip issue, append `--break-system-packages` to the end of the command, which makes it 

```
sudo pip install -r requirements.txt --break-system-packages
```

<sub>stupid pip...</sub>

Deathlord8027, 2025-04-29:
> > not enough space. Is it possible to allocate more space for this?
> 
> what size sd card are you using? minimum is 32gb
> 
> > Absoulute noob here, but it says '
> > pwnhyve.log': no such file or directory
> > when I run find
> 
> try `cat ~/pwnhyve/pwnhyve.log` if that doesn't work, try running the script manually: `sudo python3 main.py` and copy and paste the output here
> 
> > Also, when I ran the requirements command, it said that it was an external enviorment, adding on that I had to create a virtual enviorment using python 3.
> 
> this is a pip issue, append `--break-system-packages` to the end of the command, which makes it
> 
> ```
> sudo pip install -r requirements.txt --break-system-packages
> ```
> 
> stupid pip...

Deathlord8027, 2025-04-29:
I am using a 32 gigabyte micro sd card from sandisk

Deathlord8027, 2025-04-29:
> ```
> sudo pip install -r requirements.txt --break-system-packages
> ```

When I run this, there seems to be a problem. And I didn't see any errors when cloning the Pwnhyve repo
 ```
# sudo pip install -r requirements.txt --break-system-packages
ERROR: Could not open requirements file: [Errno 2] No such file or directory: 'requirements.txt'
 ```
The storage error appears when I run in in a virtual enviorment using venv 
```
┌──(root㉿raspberrypi5)-[~]
└─# cat ~/pwnhyve/pwnhyve.log
cat: /root/pwnhyve/pwnhyve.log: No such file or directory

┌──(root㉿raspberrypi5)-[~]
└─# sudo python3 main.py
python3: can't open file '/root/main.py': [Errno 2] No such file or directory
```
I think there might have been a problem when cloning the repo.

whatotter, 2025-04-30:
yeah, try following the instructions again and see if the issues still persist - seems like pwnhyve's folder doesn't exist anymore

Deathlord8027, 2025-04-30:
I think I've solved the not enough storage problem by running the following command in a virtual enviorment

 ```
mkdir -p /path/to/large/cache_dir
pip install --cache-dir=/path/to/large/cache_dir -r requirements.txt
 ```

whatotter, 2025-04-30:
pwnhyve doesn't have to be ran in a virtual environment btw, i use it without one

Deathlord8027, 2025-05-01:
```
┌──(root㉿pwnhyve)-[~/pwnhyve]
└─# sudo python3 main.py
Traceback (most recent call last):
  File "/root/pwnhyve/main.py", line 17, in <module>
    from core.utils import config
  File "/root/pwnhyve/core/utils.py", line 6, in <module>
    import toml
ModuleNotFoundError: No module named 'toml'
```

whatotter, 2025-05-01:
did you pip install requirements.txt ?

Deathlord8027, 2025-05-02:
''' 
     error: command '/usr/bin/arm-linux-gnueabihf-gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for lgpio
  Running setup.py clean for lgpio
Failed to build lgpio
ERROR: Failed to build installable wheels for some pyproject.toml based projects (lgpio)
'''

Deathlord8027, 2025-05-02:
yes, but here's the output

```
    /usr/bin/ld: cannot find -llgpio: No such file or directory
      collect2: error: ld returned 1 exit status
      error: command '/usr/bin/arm-linux-gnueabihf-gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for lgpio
  Running setup.py clean for lgpio
Failed to build lgpio
ERROR: Failed to build installable wheels for some pyproject.toml based projects (lgpio)
```

whatotter, 2025-05-02:
from the discord:

> if anyone is experiencing issues with installing lgpio, try building it manually @ https://abyz.me.uk/lg/download.html

hopefully this is the last thing you need :)

Deathlord8027, 2025-05-02:
```
  File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 226, in _enable_event_detect
    GPIO.add_event_detect(
    ~~~~~~~~~~~~~~~~~~~~~^
        self.number, self._edges,
        ^^^^^^^^^^^^^^^^^^^^^^^^^
        callback=self._call_when_changed,
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        bouncetime=self._bounce)
        ^^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: Failed to add edge detection
```
Should this be a concern?

Deathlord8027, 2025-05-02:
Nevermind, I fixed it by reinstalling rpi gpis packages. Tysm!

whatotter, 2025-05-02:
great work! there is a new commit between the opening of this issue and now, so if you'd like to update you can

Deathlord8027, 2025-05-04:
When I tired using it, I realized half the features didn't work, so I rewrote kali linux and went through the instructions. At the end, I got an error:
```
  File "/root/pwnhyve/core/displayDrivers/SH1106/config.py", line 56, in RaspberryPi
    def __init__(self,spi=spidev.SpiDev(0,0),spi_freq=40000000,rst = 27,dc = 25,bl = 18,bl_freq=1000,i2c=None):
                          ~~~~~~~~~~~~~^^^^^
FileNotFoundError: [Errno 2] No such file or directory
```

whatotter, 2025-05-04:
is spi enabled? `cat /dev | grep spidev` and make sure atleast `spidev0.0` is there, but there should be three (`spidev0.0 spidev0.1 spidev0.2`)

Deathlord8027, 2025-05-04:
Another problem is that the pwgnotchi isn't working, and the the ap scanner fails to get a key, the LOLBAS have not options when picking on it, the usb ducky mouse test doesn't work, and so much more. is this, along with the issue above, a sign of corrupted files? if so, is there any way to fix this?

Deathlord8027, 2025-05-04:
> is spi enabled? `cat /dev | grep spidev` and make sure atleast `spidev0.0` is there, but there should be three (`spidev0.0 spidev0.1 spidev0.2`)

spi is enabled, but the command just states it is a directory

whatotter, 2025-05-04:
> > is spi enabled? `cat /dev | grep spidev` and make sure atleast `spidev0.0` is there, but there should be three (`spidev0.0 spidev0.1 spidev0.2`)
> 
> spi is enabled, but the command just states it is a directory

i am stupid and meant `ls`, my apologies

whatotter, 2025-05-04:
> Another problem is that the pwgnotchi isn't working, and the the ap scanner fails to get a key, the LOLBAS have not options when picking on it, the usb ducky mouse test doesn't work, and so much [...]

pwnagotchi was broken due to nexmon drivers not working properly, AP scanner should be working but haven't tested in a bit

LOLBAS and usb ducky and such work only if you have a USB data cable connected to a computer and `/dev/hidg0` exists in pwnhyve's linux (iirc that's the directory)

Deathlord8027, 2025-05-04:
> > > is spi enabled? `cat /dev | grep spidev` and make sure atleast `spidev0.0` is there, but there should be three (`spidev0.0 spidev0.1 spidev0.2`)
> > 
> > 
> > spi is enabled, but the command just states it is a directory
> 
> i am stupid and meant `ls`, my apologies
```
└─# ls /dev | grep spidev
spidev0.0
spidev0.1
spidev0.2
```
it shows all three

whatotter, 2025-05-04:
okay, so the display should work - do you still get the same error?

Deathlord8027, 2025-05-04:
The display works fine, and for some reason, whenever I restart, it shows that the log isn't a directory or file.

whatotter, 2025-05-04:
weird, i'll look into that - have you tried the newest branch/commit? just double checking

Deathlord8027, 2025-05-05:
yes... Even tried using etcher and noobs instead of the pi imager, still didn't seem to work. Maybe its a problem on my end? im currently trying out p4wnp1 aloha to see if that works...

whatotter, 2025-05-05:
I'll make an iso soon that you can flash onto your pi that i know works, are you on a pi02 or a pi0?

Deathlord8027, 2025-05-05:
pi 02 w

Deathlord8027, 2025-05-07:
I seemed to have solved the problem with a 64gb sd card... not sure if this is a problem with the installation or the card, but the LOLABs seem to be showing up now! thx for ur help!