whatotter/pwnhyve #68: pwnagotchi does not work

notchapplez, 2025-09-06:
Pwnagotchi does not start, no error just does not open.
Expected behavior?

whatotter, 2025-09-06:
uhh you're not supposed to be running the plugin manually...

check logs (`cat pwnhyve.log`) and copy the output and paste it here, it should say something

notchapplez, 2025-09-06:
[image]

Good luck.

notchapplez, 2025-09-06:
can't rly copy n paste rn, sorry

t3kkm0tt, 2025-09-06:
did u find a fix?

whatotter, 2025-09-07:
@notchapplez try the newest commit on https://github.com/whatotter/pwnhyve/commit/c159b4a4b6359c7e23c1774c7d605033130ae363, branch `pr-2`

do be warned that it may not work correctly since nexmon is weird, sometimes it'll deauth, sometimes it wont. aswell as this plugin being real old and deauthing being completely useless nowadays (remember, pwnagotchi was released in 2019)

notchapplez, 2025-09-07:
great 
I don't have the pwnhyve at the moment, @t3kkm0tt does.

notchapplez, 2025-09-07:
We kinda share it

t3kkm0tt, 2025-09-07:
@whatotter sorry but there are still issues 

```
/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\
[PWNAGOTCHI] whitelist:
[PWNAGOTCHI] interface: wlan0
Exception in thread Thread-13 (pwnagotchi):
Traceback (most recent call last):
  File "/usr/lib/python3.13/threading.py", line 1043, in _bootstrap_inner
    self.run()
    ~~~~~~~~^^
  File "/usr/lib/python3.13/threading.py", line 994, in run
    self._target(*self._args, **self._kwargs)
    ~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/pwnhyve/addons/plugins/WiFi/wifi.py", line 126, in pwnagotchi
    cli = bcap.Client(iface=interface)
TypeError: Client.__init__() got an unexpected keyword argument 'iface'
/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\

whatotter, 2025-09-08:
try https://github.com/whatotter/pwnhyve/commit/d27cb23bb5f1a991050b0cf007fcc651961d45c0 @t3kkm0tt

whatotter, 2025-09-15:
then again, pwnhyve's pwnagotchi implementation has been buggy and probably will be since i've moved on to supporting "better" attacks like evil twin and such

notchapplez, 2025-09-15:
Okay